sfstar/hass-victron #217: Two batteries - Integration only one showing

Levanterman, 2024-06-30:
Hi

Great integration, love it, but slight problem
I have two batteries , and although the SoC and Soc2 are showing, they are identical, you can see the screen shots of the Cerbo showing two batteries, but this integration showing two identical readings.
I also have 3xMPPT but can only show two, minor.
Thanks

[image]
[image]
[image]

sfstar, 2024-07-02:
Thank you for opening this issue.
Could you perhaps share the unit id's of the batteries that are connected in your system?
You can find them under:
"Settings -> services -> Modbus TCP -> available services"

Levanterman, 2024-07-02:
> Thank you for opening this issue. Could you perhaps share the unit id's of the batteries that are connected in your system? You can find them under: "Settings -> services -> Modbus TCP -> available services"

Thanks for the reply, I went to Available Services and now I can see why the second battery is not showing.
[image]

only 1 battery is being publlished by Modbus, and then i remembered that when you add the second battery to the system you do not connect the second battery to the BMS-CAN, here is a snippet from the Drift Pro Manual:

[image]

So both batteries show up in the Cerbo but one is connected to the BMS-CAN and the other to VE-CAN

[image]

So as well as Modbus I enabled MQTT, and I can see the other battery on the Cerbo MQTT
[image]

so i added entities in the configuration.yaml from the MQTT and used those entities for the second battery, Soc, Volt, Current and you can see in the graphical integration

[image]

Looks like the instanceID might be 512, but its not being published by Modbus so maybe not accessible by your integration, but thank you , it great.

closed 2024-07-05 by Levanterman