sfstar/hass-victron #220: Overflow of sensor.victron_solarcharger_yield_user_X

dummy74, 2024-07-11:
On one of my "MPPT RS 450/100", the value for the total energy 'sensor.victron_solarcharger_yield_user_1' (register 790) started at zero again a few days ago. The overflow took place at 6553.5 kWh, which in my opinion looks like an incorrect data type for this register.
[image]

The remote console of my Cerbo shows me the correct value for the aforementioned solar charger under 'System Yield' and 'Total Yield', which is now 6635.27 kWh.
[image]

sfstar, 2024-07-23:
Hello and thank you for opening this issue.
I believe victron added new registers with a larger datatype containing the exact same information to prevent this issue.
introduced in one of the 3.x releases and slated to be released when https://github.com/sfstar/hass-victron/pull/174 is landed

klixx23, 2024-08-05:
Hello everybody,

for 3 week i have the same problem with my MPPT RS 450/100.

adrian-schutte, 2024-08-12:
Hi,

I seem to have a similar issue but instead on the "battery history chargedenergy" and "battery history dischargedenergy" sensors for about a day now. These sensor values are not updating and therefore the "Energy" Dashboard is not showing any energy going into and out of the battery any more.

[image]

dummy74, 2024-10-24:
> Hi,
> 
> I seem to have a similar issue but instead on the "battery history chargedenergy" and "battery history dischargedenergy" sensors for about a day now. These sensor values are not updating and therefore the "Energy" Dashboard is not showing any energy going into and out of the battery any more.
> <img alt="image" width="124" src="https://private-user-images.githubusercontent.com/85107932/357120299-8489179f-735e-45d9-b00f-463a0ed346a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3ODgzMTYsIm5iZiI6MTcyOTc4ODAxNiwicGF0aCI6Ii84NTEwNzkzMi8zNTcxMjAyOTktODQ4OTE3OWYtNzM1ZS00NWQ5LWIwMGYtNDYzYTBlZDM0NmE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI0VDE2NDAxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQyZTk4ZmZkMDU1MGU3MjZlMDE1NjI3YmY2MWY0ZWExZTQzZjcxYjNhNWMyNmEyNjA0ZDIzODk1NzE4NzBlMGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.rqOkAtyvVfxGJA3xQQ1Cdt8oAWXcDZKHPzefHeBAZMA">

This is normal when using CAN bus batteries (e.g. Pylontech) with an external BMS.

closed 2025-03-05 by github-actions[bot]