sfstar/hass-victron #222: "settings" imported twice with different model IDs

DSan86, 2024-07-18:
Hello, I am using this integration for one day only, but I realized that the device "settings" is imported twice to my HA with all the sensors doubled (they appended an "_2" to every sensor)
All sensors show same values and are update simultaniously. Can I safely delete one of the devices with all of the entities? If yes, which device should I delete? Modell 0 or Model 100?
Thanks a lot
[image]

[image]

sfstar, 2024-07-25:
Hello and thank you for opening this issue,

Model 0 should not contain settings entities.
You can safely disable/delete those (leave other model 0 entities untouched however).
Will need to further investigate why model 0 with settings entities is even being created

closed 2025-03-05 by github-actions[bot]

Head, 2025-03-20:
Same for me. Sadly the model 100 is the one with all the "_2" sensors, so I've disabled the model 100. I can not be deleted, only disabled.