sfstar/hass-victron #224: [Feature request] Disable non active or  unnecessary entities

lucode, 2024-07-24:
I just installed this integration and it seems to work quite well. But it burdens the recorder with many unnecessary entities.
The total in my setup is more than 200 entities.

I can identify possible ways to cut the number down:

1. When I have a Single phase setup all L2 and L3 entities are not necessary, not active.
2. The Victron modbus protocol has many doublettes, which means several identical values can be received from different devices. Could be tricky to identify those, I guess you are creating those values by scanning modbus, so to identify doublettes must be done after the scan. Ev. you can make it possible to mark them as doublettes afterwards. Plus a list of known doublettes can be managed by the community and integrated into the scan process. 
3. Another method to decrease the impact on the recorder, could be to have a built-in switch for the recorder. By default, only the entities which are essential would have the recorder switched on. Dissable them switches the recorded off too I guess.

Apfuntimes, 2024-08-05:
To make it easier, I select all of my entities and hide them, through the bulk select feature in home assistant.

I then manually select the 10 or so I want to see… I may make helpers and hide all of the native ones so when I have to change the ip address and delete and re-add the hub, I don’t have to do it all again.

closed 2024-10-07 by lucode

lucode, 2024-10-07:
https://community.home-assistant.io/t/bulk-edit-for-entities/289593/9
Didn't know about so far.