sfstar/hass-victron #225: Strange behavior but not sure where the problem occurs

sstolk90, 2024-08-01:
I'm using this integration to read my Victron values into Home Assistant.
I'm using a Node-Red flow to insert the data into the Victron GX device to report the amount of solar of my complete system.

This is sometimes happening:
[image]

And the result is this in HA:
[image]

Not knowing where to start I'm posting this here but it can that this HACS solution is not doing anything wrong.

sstolk90, 2024-08-01:
And this is haapinging almost every few days:
[image]

sstolk90, 2024-08-01:
I will monitor in the next few days with edited code from Node-Red to exclude some settings and values.

bierbuik, 2024-10-13:
I can confirm it is the latency of the response.
Problem is solved here by setting HA to check fewer.

sfstar, 2024-12-11:
Great to read that the cause has been found.
Unfortunately the gx device sometimes can't query/generate the response for an register read quickly enough.
Thereby, failing the request and causing the data to be poluted by 0 values. 
Reducing the polling frequency and improving wifi connection stability/hard wiring the devices where possible reduces the amount of times this issue occurs.
The reason why you are seeing the random spikes in the statistics data of HA is due to a bug with the resetting of statistics data when a request fails and return an 0 value.
This causes the next datapoint for statistics generation to be based on the 0 value and thereby causing an massive spike of sudden solar energy being generated while the actual sequence could be 5000 - 0 - 5001 (resulting in a 5001 spike in the graph)

sstolk90, 2024-12-12:
Makes sense! I will try to increase the interval of the HA towards Victron

closed 2024-12-12 by sstolk90