sfstar/hass-victron #233: 2 Victron Multiplus II "Input current limit"

txubelaxu, 2024-09-22:
I have 2 Multiplus II working in parallel (master and slave). And a Cerbo GX:

[image]

[image]

Using Home Assistant rule:
If I try to set a "vebus activein currentlimit" up to 20.0 and there is no problem.
But if I try to set a "vebus active in currentlimit" to 25.5, for example, I reach a [-20,20] limit:

[image]

How could I set 25.5?

Many thanks!!!

mat-l, 2024-11-04:
I am having a similar issue, just that I am able to change the value at all from within Homeassistant.
I am able to set the current limit from Victron Energys VRM app e.g. from 16 to 50 A, and the changed values are visible in Homeassistant, but I cannot use the slider or a automation to adjust the current limit:
[image]

closed 2025-03-05 by github-actions[bot]

txubelaxu, 2025-03-09:
Is a bug or something that must work this way for some reason?

Many thanks

txubelaxu, 2026-08-20:
Found the actual root cause for this on my own system (2x MultiPlus-II in parallel, same setup as described above).

It's not a Victron-side current-limit issue — VE.Bus via a GX/remote panel already treats the input current limit as the aggregated total per phase (per Victron's own manual, *Parallel, split- and three-phase VE.Bus systems*: "Setting 30A in a three phase system of six units (two per phase), on a DMC or GX Device, results in a max input current limit of 30A per phase"), so no manual multiplication by unit count is needed there.

The actual cause is this integration's own advanced config option (`ac_current`, set during initial setup, `custom_components/victron/number.py`). `number.victron_vebus_activein_currentlimit_XXX` is an AMPERE-unit register, and `determine_min_value`/`determine_max_value` use that option **directly** as ±min/max for ampere-type sliders (no voltage/phase multiplication, unlike the watt-based ones which do `V × phases × A`). If `ac_current` was left at its default/placeholder value (in my case it was `1`) instead of the real system current limit, every ampere-based slider — including this one — gets clamped to that placeholder, regardless of what the device itself actually supports.

I confirmed this empirically by sweeping the `ac_current` option through several values and reading back the resulting min/max on both an ampere-type entity and a watt-type entity:

| `ac_current` set | `vebus_activein_currentlimit` range (A) | `ess_acpowersetpoint` range (W) |
|---|---|---|
| 1 | -1 / 1 | -200 / 200 |
| 5 | -5 / 5 | -1200 / 1200 |
| 10 | -10 / 10 | -2300 / 2300 |
| 16 | -16 / 16 | -3700 / 3700 |
| 20 | -20 / 20 | -4600 / 4600 |
| 30 | -30 / 30 | -6900 / 6900 |
| 35 | -35 / 35 | -8000 / 8000 |
| 50 | -50 / 50 | -11500 / 11500 |
| 63 | -63 / 63 | -14500 / 14500 |

The ampere entity's range is always exactly ±`ac_current` (1:1, no scaling); the watt entity's range is `round(V × phases × ac_current / 100) × 100` — hence 25.5A never being reachable if `ac_current` was left at 20 (which matches the `[-20,20]` clamp reported above).

**Fix:** reconfigure the integration (Settings → Devices & Services → Victron GX modbus TCP → Configure) and set `ac_current` (and `dc_current`) to your system's real current limits — you can read the real value first from `number.victron_vebus_activein_currentlimit_XXX` itself if it was ever set correctly via VictronConnect/the GX, since that's independent of this option. Once corrected, every affected slider's min/max updates immediately (config entry reload, no HA restart needed), and setting e.g. 25.5A within the real range works fine.

Might be worth surfacing this more clearly in the docs — it's an easy trap to fall into (silently wrong default) and this thread got closed without the actual explanation.