sfstar/hass-victron #237: settings ess max dischargepower

Alexander1714, 2024-09-29:
There is obviously a max limit of 1700 Watt which i can not explain.
[image]
Venus OS show this 1700 Watt. But in Venus OS it is possible to set up to 3500Watt.
HA core is 2024.9.3 OS: 13.1
hass-victron: 0.0.11

dougle03, 2024-10-11:
I'm having the same problem. Have you tried removing the integration and re-adding it? I think it takes the min/max from the Cerbo at time of setup and does not update when you make changes using VEConfig.

Alexander1714, 2024-10-14:
Hi Peter, no i didn`t try to remove the integration. I´m a little bit scared of it. I don´t think there was a min/max during the installation of the venus os. It doesn´t matter which limit is set for maxdischargepower. It is always 1700W in the venus os.

Alexander1714, 2024-10-18:
I changed the language from German to English, then the limit of the inverter power shows again1700 W. So Peter, you are maybe wright. This strange powerlimit of 1700 W is a setting in the venus os. Do you have any idea how to change this setting?

dougle03, 2024-10-27:
Turns out that it's related to the configuration in HA when adding the integration. You can change it by going back to the HA integration, pay attention to the Amps entry.
[image]

Alexander1714, 2024-10-28:
OMG, i´am an idiot. Thank you so much for your help Peter. It works perfect now.

closed 2024-10-28 by Alexander1714

Alexander1714, 2024-10-28:
I had a wrong **DC current limit setting** in HA. I´ve change it from 35 to 55 A wich is now close to 2600 W for  the inverter.

dougle03, 2024-10-28:
No worries, I had the same problem.