sfstar/hass-victron #239: Victron integration no more recognizes any devices, everything is listed as unavailable.

martan1912, 2024-09-30:
Something happened and my integration doesn't work anymore, not one Victron device is recognized. I recently updated firmware on all 4 devices, 3 solar charger controllers and 1 shunt. Not sure this is related.
I'm running Raspberry PI version 3.42 firmware and it looks normal, just HA integration is not seeing anything.

[image]
[image]
[image]
[image]
[image]

lucode, 2024-10-02:
Same here after updating the Cerbo to v3. 42. But connection seems to be OK. But values last updated hours ago.

datraveller, 2024-11-22:
I have been having similar issues, eventually I deleted the Integration via Hacs and tried re adding it. However, now when I try add the Victron integration now it creates errors and HA restarts.

Supervisor logs:
2024-11-22 17:37:22.578 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2024-11-22 17:38:23.583 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2024-11-22 17:38:23.583 WARNING (MainThread) [supervisor.misc.tasks] Watchdog missed an Home Assistant Core API response.
2024-11-22 17:39:03.590 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2024-11-22 17:40:54.569 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2024-11-22 17:40:54.570 ERROR (MainThread) [supervisor.misc.tasks] Watchdog missed 2 Home Assistant Core API responses in a row. Restarting Home Assistant Core!
2024-11-22 17:40:54.583 INFO (SyncWorker_3) [supervisor.docker.manager] Restarting homeassistant
2024-11-22 17:41:44.559 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.

closed 2025-03-05 by github-actions[bot]