sfstar/hass-victron #245: No module named 'pymodbus.register_read_message'

usersno, 2024-10-11:
Hi, 
I'm am running the victron plugin v0.3.0 on the current home assistant 2024.10.2, supervisor 2024.10.0 and operating system 13.1. 
Since yesterday the victron plugin does not load anymore. Not sure if that correlates with the last supervisor update or something else. Does anybody have any ideas what und where to fix?

The log shows the following error:

Logger: homeassistant.setup
Source: setup.py:334
First occurred: 21:56:42 (1 occurrences)
Last logged: 21:56:42

Setup failed for custom integration 'victron': Unable to import component: No module named 'pymodbus.register_read_message'

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1007, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1067, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/victron/__init__.py", line 9, in <module>
    from .coordinator import victronEnergyDeviceUpdateCoordinator as Coordinator
  File "/config/custom_components/victron/coordinator.py", line 14, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

speckboy64, 2024-10-19:
I had the same problem as you. I installed the HA update yesterday and it looks like it solved the problem. I hope that is the same case for you. :)

hackex, 2024-10-23:
I have the same error message with all the newest updates from HASS and Victron add-on.
Any fixes?

speckboy64, 2024-10-26:
For sum reason the problem reappeared in my case. Here is my log output:

2024-10-26 20:10:28.358 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'victron': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1007, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1067, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/victron/__init__.py", line 9, in <module>
    from .coordinator import victronEnergyDeviceUpdateCoordinator as Coordinator
  File "/config/custom_components/victron/coordinator.py", line 14, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

Wixely, 2024-10-27:
I updated to HA 2024.10.4 and I'm getting this issue too, the whole integration has stopped working for me.

speckboy64, 2024-11-03:
Hello. I did a little research and I found a similar problem in the huawei_solar integration: https://github.com/wlcrs/huawei_solar/issues/670#issuecomment-2031136245

The summary is that other integrations also use pymodbus however in a different version, therefore the version needed for this integration is not available. This might answer the question why not everybody has this issue.

Wixely, 2024-11-07:
I downgraded to the version in the hass-victron manifest

```pip install pymodbus==3.5.1```

Worked after a restart. I did have a Solarman integration installed which I removed but that alone didnt resolve it. I was stuck in a state of being unable to remove or reinstall the victron integration. The version I downgraded from was pymodbus 3.7.4

usersno, 2024-11-07:
With today's update to Home Assistant 2014.11.0 the issue is resolved for me as well. I hope it stays that way and does not resurface again.

speckboy64, 2024-11-07:
The update to Home Assistant 2014.11.0 did not solve it for me. The probem is still present.

usersno, 2024-11-08:
Unfortunately, for me it's back again :-(

usersno, 2024-12-24:
I found the problem and solution for my environment: The plugin uses modbus, which is part of Home Assistant. Unfortunately, it does not initiate/load modbus correctly. If modbus is already present due to a separate config / use case, everything works fine. If there is no other modbus config, most of the time modbus isn't loaded and subsequently loading the victron plugin fails.

speckboy64, 2024-12-30:
Ok, so how did you solve the problem in your system? I have another plugin which uses modbus but in my case it still does not work.

usersno, 2024-12-31:
I added a manual modbus connection to the same victron device via modbus, but only configured one sensor. That ensures that modbus is loaded and can create the connection. The rest then is successfully handled by the victron plugin.

bwurst, 2024-12-31:
I fixed the same issue by changing the version in requirements in victron module to "pymodbus==3.5.4".
In my case, homeassistant has uninstalled the pip version (because it's bundled one satisfied the ">=3.5") and victron module failed with that.

usersno, 2025-01-04:
This workaround helps as well:
https://github.com/sfstar/hass-victron/issues/267#issuecomment-2569940266

dustball62, 2025-01-08:
apparently it must be exactly  "pymodbus==3.5.4" not  "pymodbus>=3.5.4"  or anything else.