sfstar/hass-victron #259: HA Fehler: config_flow.py, line 257: self.config_entry = config_entry

crisi-solar, 2024-12-10:
Detected that custom integration 'victron' sets option flow config_entry explicitly, which is deprecated at custom_components/victron/config_flow.py, line 257: self.config_entry = config_entry. This will stop working in Home Assistant 2025.12, please create a bug report at https://github.com/sfstar/hass-victron/issues

Der Fehler, den du beschreibst, befindet sich in der benutzerdefinierten Integration victron für Home Assistant, genauer gesagt in der Datei config_flow.py in Zeile 257. Dort wird die Option config_entry explizit gesetzt, was in zukünftigen Versionen von Home Assistant (ab 2025.12) nicht mehr unterstützt wird.

sfstar, 2024-12-11:
Thank you for opening this issue and  bringing this to my attention.
I will look into changing the configflow to remove the deprecation.

closed 2025-05-05 by github-actions[bot]

MatthiasOs, 2025-07-24:
Sill relevant, I received the Issue today.