sfstar/hass-victron #263: 2025.1 issue

remcom, 2024-12-23:
The new home assistant beta results in the following error

```
Logger: homeassistant.setup
Source: setup.py:334
First occurred: 20:13:36 (1 occurrences)
Last logged: 20:13:36

Setup failed for custom integration 'victron': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/victron/__init__.py", line 9, in <module>
    from .coordinator import victronEnergyDeviceUpdateCoordinator as Coordinator
  File "/config/custom_components/victron/coordinator.py", line 14, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'
```

dalewright, 2024-12-26:
Likewise. 

```
Logger: homeassistant.setup
Source: setup.py:334
First occurred: 3:07:14 PM (1 occurrences)
Last logged: 3:07:14 PM

Setup failed for custom integration 'victron': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/victron/__init__.py", line 9, in <module>
    from .coordinator import victronEnergyDeviceUpdateCoordinator as Coordinator
  File "/config/custom_components/victron/coordinator.py", line 14, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

```

devilmastah, 2025-01-01:
I have the same issue here

kasper334, 2025-01-03:
Just updated to 2025.1.0, got same error.
Thanks a lot, @devilmastah, got it up and running.

kasper334, 2025-01-03:
@andreas-bulling not sure if my solution is correct (i highly suspecting it is not), but anyways it's working for me.
I fixed mine by reinstalling "over" an updated repo.

Here are my steps:
1. Copy a link to a forked repo, which fixes this error (for example, I took `https://github.com/julesxxl/hass-victron`).
2. Open Home Assistant, go to HACS
3. Press three dots at the top right corner, click on "Custom repositories".
4. Paste copied link into "Repository" field, then choose "Integration" in the "Type" field.
5. Click "Add".
6. Go back to HACS initial page, search for "Victron" integration in the search box, click on the result without description (it just says "Victron GX modbus TCP" in a single line).
7. Click "Download" on the bottom right corner, this will overwrite contents in `custom_components/victron` path.
8. Restart Home Assistant.

andreas-bulling, 2025-01-03:
Thanks! I ended up doing a manual install and directly using the version from @remcom repository.

ChuckRamsey, 2025-01-03:
[kasper334](https://github.com/kasper334) Your solution worked fine for me. Thanx!

tmonxx, 2025-01-03:
@kasper334 Thx. Now it works 💯

asmcc, 2025-01-04:
In [#267](https://github.com/sfstar/hass-victron/issues/267#issuecomment-2569940266) you can find an alternatively [workaround](https://github.com/sfstar/hass-victron/issues/267#issuecomment-2569940266) WITHOUT forked repository.

pos-ei-don, 2025-01-04:
Maybe we should add some or the other prs (corewhtion of ah instead of a,...) also into the fork, until the original maintainer can uptahte his rwpositoeye again (or Hand it over to a new maintainer?)

aliasimac, 2025-01-04:
Worked, thanks

QaDoPL, 2025-01-06:
> @andreas-bulling not sure if my solution is correct (I'm highly doubtful), but anyways it's working for me. I fixed mine by reinstalling "over" an updated repo.
> 
> Here are my steps:
> 
>     1. Copy a link to a forked repo, which fixes this error (for example, I took `https://github.com/julesxxl/hass-victron`).
> 
>     2. Open Home Assistant, go to HACS
> 
>     3. Press three dots at the top right corner, click on "Custom repositories".
> 
>     4. Paste copied link into "Repository" field, then choose "Integration" in the "Type" field.
> 
>     5. Click "Add".
> 
>     6. Go back to HACS initial page, search for "Victron" integration in the search box, click on the result without description (it just says "Victron GX modbus TCP" in a single line).
> 
>     7. Click "Download" on the bottom right corner, this will overwrite contents in `custom_components/victron` path.
> 
>     8. Restart Home Assistant.
> 
> 
> If someone will decide to repeat my steps - please don't forget to remove second repo from HACS (which was added in steps 1-5) in future, when owner of the current repo will merge this fix and update will be available through HACS.

This solve issue, but not all entities are visible ex can't control Grid power with ESS.

QaDoPL, 2025-01-06:
number.victron_settings_ess_acpowersetpoint and others control entities solved by again download data from @sfstar [repo](https://github.com/sfstar/hass-victron)  and now everything works like a harm.

Regards

etofi, 2025-01-06:
> In short: download @julesxxl hass-victron repo, if everything start then download again sfstar and voilà Regards

If I do this, the Victron addon does not start. As the error still exists in the sfstar code

closed 2025-04-07 by github-actions[bot]