sfstar/hass-victron #267: Not work after upgrade to Home Assistant 2025.1.0

danygi79, 2025-01-03:
hello, after upgrade to 2025.1.0 this integration stop work.

the log is:

Registratore: homeassistant.setup
Fonte: setup.py:334
Prima occorrenza: 23:38:13 (1 occorrenze)
Ultimo accesso: 23:38:13
Setup failed for custom integration 'victron': Unable to import component: No module named 'pymodbus.register_read_message'

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/victron/__init__.py", line 9, in <module>
    from .coordinator import victronEnergyDeviceUpdateCoordinator as Coordinator
  File "/config/custom_components/victron/coordinator.py", line 14, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

tmonxx, 2025-01-03:
Hi
Same problem...

weide-git, 2025-01-03:
Same to me:

Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 23:15:18 (1 Vorkommnisse)
Zuletzt protokolliert: 23:15:18

Setup failed for custom integration 'victron': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/victron/__init__.py", line 9, in <module>
    from .coordinator import victronEnergyDeviceUpdateCoordinator as Coordinator
  File "/config/custom_components/victron/coordinator.py", line 14, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

neil192, 2025-01-03:
I am also having the same issue, i have had to revert back

BMG80, 2025-01-03:
Same problem here

BMG80, 2025-01-03:
If you’re encountering the error:

ModuleNotFoundError: No module named 'pymodbus.register_read_message'

Here’s a potential fix:
	1.	Open the file  /config/custom_components/victron/coordinator.py.
	2.	Comment out line 14, which imports ReadHoldingRegistersResponse:

# from pymodbus.register_read_message import ReadHoldingRegistersResponse

After making this change, everything seems to work fine for me.

asmcc, 2025-01-04:
Yes, it works for me too as workaround. But the big question here is: Does the intergation work 100% without this line and without Holding Registers? Since generally Holding Registers are very important, if you use ModBus. And I do not know, if the intergation really needs this line or not.

928stuart, 2025-01-04:
I have same sort of issues.
On trying to start Victron I get - Config flow could not be loaded: {"message":"Invalid handler specified"}
The potential fix didn't work for me

drRax, 2025-01-04:
This issue is an exact duplicate of https://github.com/sfstar/hass-victron/issues/263

asmcc, 2025-01-04:
Yes, but the workaround solution from here should be publicated in #263 too, since the solution from #263 with forked repository and manual change first to this fork and thereafter back to the "original" repository is not really a good idea. The proposed workaround with comment out of one line is better. Since, if the "original" HACS will be fixed in the next future, you do not need this disclaimer "please not forget to change back to the original repository"

iJAY86, 2025-01-04:
Thanks @BMG80

johnyhravy, 2025-01-04:
Hello. It worked perfectly for me too. Thank you very much. @BMG80

asmcc, 2025-01-04:
Please mark this issue as duplicate, close it and let us continue the discussion in #263

Marjaletes, 2025-01-04:
funciona con actualizacion Home Assistant
#BMG80 
He aquí una posible solución:

Abra el archivo /config/custom_components/victron/coordinator.py.

Comente la línea 14, que importa ReadHoldingRegistersResponse:
# from pymodbus.register_read_message import ReadHoldingRegistersResponse

Reiniciar Home Assistant.

Después de realizar este cambio y reiniciar Home Assistant, todo parece funcionar bien para mí.👍👍

chasbrook, 2025-01-04:
> 1. Open the file /config/custom_components/victron/coordinator.py.
> 2. Comment out line 14, which imports ReadHoldingRegistersResponse:
>    `# from pymodbus.register_read_message import ReadHoldingRegistersResponse`
> 3. Restart Home Assistant.

**_Thanks! Worked fine ..._**

asmcc, 2025-01-04:
Can someone close this issue as duplicate, please!!!
And no comments please here!!! That is a DUBLICATE. If you have any constructive ideas or further proposals, then under #263, but not here, please! And yes, we know, that this workaround works and we do not need here further feedback about it, that it works!

flipponweb, 2025-01-05:
> Here’s a potential fix:
> 
> 1. Open the file /config/custom_components/victron/coordinator.py.
> 2. Comment out line 14, which imports ReadHoldingRegistersResponse:
>    `# from pymodbus.register_read_message import ReadHoldingRegistersResponse`
> 3. Restart Home Assistant.
> 
> After making this change and restarting Home Assistant, everything seems to work fine for me.

works for me, too . Good job !

msevestre31, 2025-01-05:
Also for me (done yesterday). However, there might be some hidden unwanted side effects i'm not confortable with - thus, hope for an official fix

vdneves, 2025-01-05:
Tha

> Here’s a potential fix:
>
> 1. Open the file /config/custom_components/victron/coordinator.py.
> 2. Comment out line 14, which imports ReadHoldingRegistersResponse:
>    `# from [...]

Muito Obrigado... Excelent.

remcom, 2025-01-05:
Hi all! Because this repo is maintained and i also like to use my Victron equipment in HA i made a new repo that you can use: https://github.com/remcom/hass-victron

My goals: 
- process a lot of the outstanding pr's on this repo
- have a look at the open issues here
- code cleanup
- Adding support for the new Victron features that are added in the last months

Feel free to make your own pr's on the new branch or make issues on stuff you running into. Also if there are fellow maintainers that want to join please send me a message

Cirion75, 2025-01-11:
Worked! Thanks!

aihngel, 2025-01-12:
Switched to @remcom 's repo and it's working great!  Thanks!  Easy instructions on how to switch too btw.

Marjaletes, 2025-01-12:
Ok, thanksEnviado desde Outlook para AndroidDe: aihngel ***@***.***>Enviado: domingo, enero 12, 2025 3:15:55 a. m.Para: sfstar/hass-victron ***@***.***>CC: Marjaletes ***@***.***>; Comment ***@***.***>Asunto: Re: [sfstar/hass-victron] Not work after upgrade to Home Assistant 2025.1.0 (Issue #267)
Switched to @remcom 's repo and it's working great!  Thanks!  Easy instructions on how to switch too btw.

sfstar, 2025-01-27:
Hey Guys,

Sorry for my absence the last period.
Life was getting in the way of maintaining this repository.
I see that remcon has created an working fork with the necessary patches for version 2025.1 and some other improvements. (Great job 👍 ).

I see now that a project of this size without any additional maintainers to approve pr's and review code isn't an workable solution.
I've contacted Remco to discus the future of this project so that the communities development effort and the stability for users can be better guaranteed.

Regards,

Sfstar

remcom, 2025-01-27:
Hee @sfstar. Good to see you back! I Didn't see your message (yet). Dont know where you send it to?

sfstar, 2025-01-28:
Hello Guys,

I've had contact with remcom via chat.
Since the availability of quick updates as well as an improved code quality is important we discussed adding him and @julesxxl to the repository as maintainer so that they as well can review/merge PR's.
Also we will be working on getting @remcom his changes back into this repository.
once this is stable, users that are currently on the fork can choose to migrate back to this repository.

Hopefully this will ensure that you guys as the community get faster support/responsetimes.

Will leave this issue open until the ported release is available.

sfstar, 2025-02-03:
The port has been merged to master and will be included in the next release.
This issue will remain open until then.
Regarding compatibilty with HA core, thank to the port of changes made by @remcom v0.4.2 is now fully prepared for HA core 2025.2.
#281 will be expanded upon to prepare this integration for such a time when HA core switches over to pymodbus 3.9.0 or higher (which will break the integration due to the decoding methods being deprecated, to be removed in pymodbus 3.9.0).

TrippleDelta, 2025-02-07:
> Here’s a potential fix:
>
>     1. Open the file /config/custom_components/victron/coordinator.py.
>
>     2. Comment out line 14, which imports ReadHoldingRegistersResponse:
>        `# from [...]

Since Core 2025.2.0 this fix doesn't work anymore.

TrippleDelta, 2025-02-07:
> Since Core 2025.2.0 this fix doesn't work anymore.

Sorry, v0.3.5 fixed it. Everything works fine now.

sfstar, 2025-03-05:
With the release of 0.0.5 changes have been backported and compatibility with 2025.1 and 2025.2 has been created.
Closing this issue as resolved.
Feel free to re-open if it persists after upgrading to the latest hass-victron release

closed 2025-03-05 by sfstar