sfstar/hass-victron #268: Not working after update to 2025.1.0

penguinator13, 2025-01-04:
Victron energy integration broke after updating home assistant up to 2025.1.0. Removed integration and tried adding again but get the message - "Config flow could not be loaded: {"message":"Invalid handler specified"}".
Thanks.

Newevg, 2025-01-04:
Same for me

mgies, 2025-01-05:
same here. Got an error stating pymodbus was missing.. was it somehow changed/updates with homeassistant 2025.1.0?

podilamaya, 2025-01-05:
same error for me too

cfish5, 2025-01-05:
yep same here....with so many integrations tied to Victron as we live off grid, very keen to find a solution !!!

Busta999, 2025-01-05:
Same issue, I have fallen back to using MQTT.

ganan20, 2025-01-05:
Erreur
Le flux de configuration n'a pas pu être chargé: {"message":"Invalid handler specified"}

hondamz, 2025-01-05:
also here the same. No Data from Victron devices. 

HA Core Version is 2025.1.0 (other HA Instance with older Version is still running!)
Victron Intetragion Version: 0.0.11
Log shows this (sorry can not interpret this because I'm not an developer):

Setup failed for custom integration 'victron': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

evercape, 2025-01-05:
> also here the same. No Data from Victron devices.
>
> HA Core Version is 2025.1.0 (other HA Instance with older Version is still running!) Victron Intetragion Version: 0.0.11 Log shows this (sorry can [...]

same issue here. manually updated to HACS 2.0.2 and everything else is on the latest version HA 2025.1

marcmanusch, 2025-01-05:
Same issue here.

nervousfrog, 2025-01-05:
Same problem here. 

I managed to roll back to 2024.12.5 and back up and running.

Just in case it helps anyone from the terminal I entered 

ha core update --version 2024.12.5

Busta999, 2025-01-05:
In implementing a workaround where I drive the Victron through its MQTT Broker I noticed the path to control it had changed.

Until today it was using N/48e7da8663a9/settings/0/Settings/CGwacs/BatteryLife/MinimumSocLimit to set the SOC Limit - ie 100 sets it keep the battery at 100%, and 20 to keep it at 20% - ie force a charge or discharge.

While setting this up to link into the Octopus Mini - binary_sensor.octopus_energy_a_059ca71a_intelligent_dispatching to detect changes in Peak and Offpeak tariffs I noticed the Victron MQTT path had changed to

W/48e7da8663a9/settings/0/Settings/CGwacs/BatteryLife/MinimumSocLimit

Now is probably unrelated but might explain why the Victron Integration is failing?

For what it's worth.

RobMeerwijk, 2025-01-05:
Same error

remcom, 2025-01-05:
Hi all! Because this repo is maintained and i also like to use my Victron equipment in HA i made a new repo that you can use: https://github.com/remcom/hass-victron

My goals: 
- process a lot of the outstanding pr's on this repo
- have a look at the open issues here
- code cleanup
- Adding support for the new Victron features that are added in the last months

Feel free to make your own pr's on the new branch or make issues on stuff you running into. Also if there are fellow maintainers that want to join please send me a message

Jacke87, 2025-01-05:
> In implementing a workaround where I drive the Victron through its MQTT Broker I noticed the path to control it had changed.
>
> Until today it was using [...]

Please delete your Portal ID. 

same issue here

RobMeerwijk, 2025-01-05:
> Hi all! Because this repo is maintained and i also like to use my Victron equipment in HA i made a new repo that you can use: https://github.com/remcom/hass-victron
>
> My goals:
>
>     * process a lot [...]

@remcom Thanks a lot, I have already added your repo to my hacs-resources. It solved this issue. And great that you continue this work.

Busta999, 2025-01-05:
> > In implementing a workaround where I drive the Victron through its MQTT Broker I noticed the path to control it had changed.
> > Until today it was using N/settings/0/Settings/CGwacs/BatteryLife/MinimumSocLimit to set the SOC Limit - ie 100 sets it keep the battery at 100%, and 20 to keep it at 20% - ie force a charge or discharge.
> > While setting this up to link into the Octopus Mini - binary_sensor.octopus_energy_intelligent_dispatching to detect changes in Peak and Offpeak tariffs I noticed the Victron MQTT path had changed to
> > W/settings/0/Settings/CGwacs/BatteryLife/MinimumSocLimit
> > Now is probably unrelated but might explain why the Victron Integration is failing?
> > For what it's worth.
> 
> Please delete your Portal ID.
> 
> same issue here

Thanks can you delete from your post too

thanks

Newevg, 2025-01-06:
Fixed by installing [sfstar]. Thank you!

tazinblack, 2025-01-06:
is there a step by step instruction for dummies what has to be done before updating HA to 2025.1.0 to not run into this problem?

remcom, 2025-01-06:
@tazinblack i will make a step by step guide this evening!

tazinblack, 2025-01-06:
Thanks a lot!

remcom, 2025-01-06:
@tazinblack the readme now contains an upgrade guide

huub99, 2025-01-07:
Thanks for the effort gents, good to sea.
Is the old integration deprecated or is there an update to be expected in the near future?

remcom, 2025-01-07:
I'm not sure. I dont no the intentions of the original maintainer.

drjuliankeates, 2025-01-08:
Working for me now as well, thank you!

huub99, 2025-01-09:
I'm doubting between transitioning to this integration or to just create sensors using modbus.
Can someone explain what the benefits are of using an integration. (modbus is an integration as well I assume, however maintained by HA themselves I assume)

remcom, 2025-01-09:
@huub99 It comes down to a trade-off between convenience and dependency on an integration.

With the Modbus integration, you’ll need to figure out all the addresses and configurations yourself, but it eliminates reliance on other integrations or external contributors.

Using the Victron integration, on the other hand, offers greater ease of use but comes with dependency on the integration itself.

huub99, 2025-01-09:
You are quite right about this one ;).
"you’ll need to figure out all the addresses and configurations yourself"  the only one I cannot get to work is the min cell voltage id and max cell voltage id.

I tried this:
- name: 'Modbus Victron highest cell ID'
      slave: 225 
      address: 1310
      data_type: string
      count: 4
      scan_interval: 5
      device_class: battery

and fiddled around with different data types like uint16 and int16. However not the desired result.
Any ideas?

sfstar, 2025-01-27:
Hello Guys,

Sorry for the lack of reaction speed.
For more information see: https://github.com/sfstar/hass-victron/issues/267#issuecomment-2616593624

sfstar, 2025-02-03:
Closing issue since port has been realised and as part of cleaning up the issue backlog.
Feel free to re-open if you feel like the discussion hasn't been resolved.

closed 2025-02-03 by sfstar