sfstar/hass-victron #269: Is this code actively maintained?

tobox, 2025-01-05:
Given that sfstar has no activities since may 2024, I would like to raise the question if this code is actively maintained?

I would like to update to 2025.1 asap, because I would love to try some of the new features, but a working victron integration is crucial for me. 

Maybe some other contributor to this repository has more insights to the current situation?

luebke28, 2025-01-05:
Good Morning,
i have the same Issue.

21-Jailbraik-21, 2025-01-05:
Ja, das habe ich mich auch schon gefragt.

evercape, 2025-01-05:
same issue here

pos-ei-don, 2025-01-05:
There is even a Video from the offgrid garage on youtube where Andy had the same Problems last year.

Thema s Integration is very important for the one who use is, so we need at least fast Updates regarding the compatibility to HA.

tobox, 2025-01-05:
I just did the obvious thing - moved to  https://github.com/remcom/hass-victron 

He even offered to provice HACS integration, if sfstar does not answer.

evercape, 2025-01-05:
> I just did the obvious thing - moved to https://github.com/remcom/hass-victron

Please let me know when you get this issue fixed? You could also do a pull request to update the original integration. 

I need it badly resolved. Running 4 installations with it.

Thanks

tobox, 2025-01-05:
> > I just did the obvious thing - moved to https://github.com/remcom/hass-victron
> 
> Please let me know when you get this issue fixed? You could also do a pull request to update the original integration.
> 
> I need it badly resolved. Running 4 installations with it.

The user remcom has already created a PR. But since sfstar seems to be absent, I just moved to remcoms repo. The issue is fixed there, everything seems to be working afaik.

remcom, 2025-01-05:
Hi all! Because this repo is maintained and i also like to use my Victron equipment in HA i made a new repo that you can use: https://github.com/remcom/hass-victron

My goals: 
- process a lot of the outstanding pr's on this repo
- have a look at the open issues here
- code cleanup
- Adding support for the new Victron features that are added in the last months

Feel free to make your own pr's on the new branch or make issues on stuff you running into. Also if there are fellow maintainers that want to join please send me a message

lthieman, 2025-01-05:
I don't code but I run Home Assistant in my travel trailer with a Multiplus and Cerbo and would love to help get HA and Victron working together better.

remcom, 2025-01-05:
@lthieman Maybe its a good idea on the new repo to make an issue with the features your missing, or not working correctly for you. I'm running an ESS setup so i need input on what needs to work beter in a camper setup

asmcc, 2025-01-05:
I find it as good idea, remcom!
Please show all the PRs here since May 2024 and try to merge it in your repo, if you can. And please make in your repository an discuss area. Issues are usually "Tickets" and are not for such discussions like this one here.
Thanks!

remcom, 2025-01-05:
@asmcc Good idea! i enabled it!

luebke28, 2025-01-05:
Ich bedanke mich mal in Deutsch:
Vielen Dank für Deine schnelle Hilfe!!
Läuft wieder alles!
Super Job!!

jwillmer, 2025-01-06:
@remcom I think you should post some quick instructions on how to switch to your repository ;-)

remcom, 2025-01-06:
@jwillmer I'm planning to that this evening. Forgot to do that

JoJones79, 2025-01-06:
@remcom I am looking forward to move to your fork and update HA. 
Heel erg bedankt voor je werk.

remcom, 2025-01-06:
The [readme](https://github.com/remcom/hass-victron/blob/master/README.md) now contains an upgrade guide!

sfstar, 2025-01-27:
Hello Guys,

my apologies for the extended absences and the issues experienced in the mean-time.

Regarding the future of this repository you can read up on it here: https://github.com/sfstar/hass-victron/issues/267#issuecomment-2616593624

Once more information becomes available I will update this thread.

sfstar, 2025-02-03:
Closing issue since port has been realised and as part of cleaning up the issue backlog.
Feel free to re-open if you feel like the discussion hasn't been resolved.

closed 2025-02-03 by sfstar