sfstar/hass-victron #272: 2025.1.2 Core update breaks the VIctron HAC

willardjwilliams, 2025-01-12:


aihngel, 2025-01-12:
Can confirm 2025.1 and 2025.2 both break Victron HACS integration. Rolling back to latest 2024.x fixes it.  Is there anyone that can pickup this project?

willardjwilliams, 2025-01-12:
Yeah I rolled back.  Thanks.

It’s a great HACS!  Really love it.  Wish I could help maintain it, but it is above my head.

aihngel, 2025-01-12:
Switched to @remcom 's repo and it's working great! Easy instructions on how to switch too btw.

willardjwilliams, 2025-01-12:
I’m a newb to this.  This person has a Victron HACS?

aihngel, 2025-01-12:
You have to swap out the "repository" in hacs for theirs instead. Basically you'll see both.  Use the instructions in theirs for removing the old one and adding the new one and after a restart you'll be great. (Ps when deleting the old hacs repository dont delete your config when it asks.)

remcom, 2025-01-12:
@willardjwilliams You can find a migration guide here https://github.com/remcom/hass-victron

willardjwilliams, 2025-01-12:
Thx that worked!

sfstar, 2025-01-27:
Hello Guys,

Sorry for the lack of reaction speed.
For more information see: https://github.com/sfstar/hass-victron/issues/267#issuecomment-2616593624

sfstar, 2025-02-03:
The integration now supports 2025.1 and 2025.2 due to changes being backported from the remcom fork.
Further information regarding the port will be given in #267.
Closing this issue as resolved.
Feel free to re-open if the issue persists.

closed 2025-02-03 by sfstar