sfstar/hass-victron #283: Smartsolar device not discovered after upgrade

pipoab, 2025-02-04:
I have notice that my smartsolar can solarcharger is not being discoved by the victron integration after updating to v0.4.0, was working before

I removed the integration from HASK and re-installed again, noting working, cannot manage to see my solarcharger anymore

sfstar, 2025-02-04:
Thank you for opening this issue,

Could you see if the smart solar is listed under services -> modbus tcp -> available services? At the gx device console

pipoab, 2025-02-04:
Yes the smart solar is visible on the  GX device

Check attached picture

Regards Philippe

sfstar, 2025-02-04:
Unfortunately the attachment do no seem to be included in the comment reply as visible on github.
Could you set the integration to "enable debug logging" under the config entry and upload the debug logging after a manual rescan? (as well as the picture of the previous comment?)

pipoab, 2025-02-09:
@sfstar   here is the debug logging after manual rescan

[home-assistant_victron_2025-02-09T16-21-14.922Z.log](https://github.com/user-attachments/files/18725207/home-assistant_victron_2025-02-09T16-21-14.922Z.log)

pipoab, 2025-02-09:
@sfstar and attached the picture showing smart solar visible under services in GX device

[image]

Oinq, 2025-05-12:
I previously removed the first integration that was working fine.

Last weekend, I physically changed my setup from 3-phase to single-phase. Because of that, I also removed the older integration and installed this new one.

Now, the MPPT is no longer showing up, and I’ve lost the production stats in Home Assistant.

Does anyone have a solution?