sfstar/hass-victron #286: BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !

rosentrieb, 2025-02-06:
Logger: pymodbus.logging
Quelle: custom_components/victron/coordinator.py:117
Integration: victron (Dokumentation, Probleme)
Erstmals aufgetreten: 18:10:38 (2236 Vorkommnisse)
Zuletzt protokolliert: 18:52:43

BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

sfstar, 2025-02-06:
Thank you for reporting.
Working on a rewrite that adresses this locally.
Changes are pushed to #281 and once it is merged the integration will be 3.9.0 compatible

julesxxl, 2025-02-15:
> Thank you for reporting. Working on a rewrite that adresses this locally. Changes are pushed to [#281](https://github.com/sfstar/hass-victron/pull/281) and once it is merged the integration will be 3.9.0 compatible

Just noticed that the deprecated message appears in logs almost 10 times per second... and that grows, grows...

[image]

sfstar, 2025-02-20:
Yes unfortunately the deprecated functionality allows for easy decoding of values.
The message will occur for every update of the integration made to the venus device.
Replacing this is a fair amount of work.
Fortunately, the deprecation is still some time out and it is clear wat needs to be done in order to migrate.

beerygaz, 2025-03-02:
Is there any way to bump the log level to error so we can suppress the warning until the module upgrade is in place?

I assume given the integration w e should be able to modify an entry in the 
'''Logger:'''section 

https://www.home-assistant.io/integrations/logger/

seb585, 2025-04-02:
```
logger:
  default: warning
  logs:
    pymodbus.logging: error
```

in configuration.yaml worked for me

johnyhravy, 2025-04-27:
Hello.
Is there already a functional fix that would prevent so many warning messages?
About 70000x warnings in 1 day!
I'm starting to notice that when opening HA in a browser, my initial screen with statistical graphs has been loading for quite a long time lately. After a few days, it stops writing and I have to restart HA.
If the advice from @seb585 would help, should I add these lines to configuration.yaml via the editor after the last line? In my case, as line 13 and onwards?

[image]

Jochen754, 2025-05-27:
Is there still work done on this topic?

closed 2025-08-26 by github-actions[bot]