sfstar/hass-victron #297: TITLE: Victron Integration - Home Assistant via Modbus: Error with updates and logs

alnavasa, 2025-02-18:
**DESCRIPTION**:
- I'm using the Modbus integration for Victron in Home Assistant.
- I receive the following error: `Error processing function code 3, unit id 227, start address 259, quantity 68, src=::ffff:192.168.20.201: Error finding service with device type battery at device instance 276`.
- The entity information does not automatically refresh and I have to manually reload it to see the correct values.

**STEPS TO REPRODUCE**:
1. Configure the Victron integration in Home Assistant via Modbus.
2. Enable the Modbus/TCP service on the Victron device.
3. Check the logs in GX and observe the reported error.
4. Notice that the entity’s information is not updated unless it’s manually reloaded (sometimes)

**EXPECTED BEHAVIOR**:
- Data should be updated automatically, without the need to reload the entity.
- No error message like the one mentioned should appear.

**OBSERVED BEHAVIOR**:
- The error appears with a timestamp (e.g.): 2025-02-18 19:24:28
- It’s necessary to reload the entity to see updated values.

**ENVIRONMENT**:
- Victron firmware version: 3,54
- Home Assistant version: 

```
Core
2025.2.4
Supervisor
2025.02.1
Operating System
14.2
Frontend
20250214.0
```

**RELEVANT LOGS**:
```
ERROR “Error processing function code 3, unit id 227, start address 259, quantity 68, src=::ffff:192.168.20.201: Error finding service with device type battery at device instance 276”
```
**ADDITIONAL INFORMATION**:
- Modbus/TCP was enabled on the Victron device from the app.
- Currently, it is unknown if there is a workaround or additional configuration required to fix this.
- Any help or suggestions are welcome.

alnavasa, 2025-02-18:
[image]
[image]

alnavasa, 2025-02-18:
[image]

sfstar, 2025-02-20:
Thank you for opening this issue and providing such detail.
Unfortunately, modbus query stability and the integrations recovery of it have been reported a couple of times in the past.
Hopefully, with the information provided in this issue we can make some headway into improving the quality of connection handling/recovery.

Will investigate the issue further in the coming days based on provided input

han-alink, 2025-02-21:
Hi,
I'm seeing the same kind of behaviour. Items (example: SOC values), are not automatically updated anymore. With a rescan they are updated, but the modbus interface also generates an error at the moment of rescanning:

[image]

han-alink, 2025-02-21:
Using a modbus.yaml entry, the soc is updated normally:
- 
- type: tcp
#  host: 192.168.1.73
  host: 192.168.1.73
  port: 502
  sensors:  
    - name: Victron SOC
      unit_of_measurement: "%"
      slave: 228 #HUB
      address: 30
      data_type: uint16
      scan_interval: 10
      scale: 0.1

spacecabbie, 2025-02-22:
Hi all just wanted to chip in here I was looking into same issue on my setup for troubleshooting I like to add that if you leave it be for an hour or so the error goes away sometimes i have to restart home assistant. But it looks like this (in my case) only occurs afther a venus os or gx device reboot.

sfstar, 2025-02-24:
Thanks all for providing information regarding the issues experienced.
For now you could use the automation in this comment (from a previous issue) to "fix" the integration until the next hickup.
https://github.com/sfstar/hass-victron/issues/64#issuecomment-1488023571

I'm looking at an partial rewrite for the connection logic to make this more robust (currently the retry logic has some bugs in it).
Due to the amount of work required this might take 1 or 2 weeks.
If you notice any other behaviour related to this issue by the integration (feel free to comment on this issue.)

alnavasa, 2025-02-25:
Hi, i dont want to kill @sfstar work, but I moved to mqtt (many guides out there) and it works reliable with updates any second. maybe going ModBus is an overkill of hard work to make this out, or maybe it is a real solution for some people that want to move away from mqtt.

alnavasa, 2025-02-25:
Maybe a proper mqtt integration can be write up for this.