sfstar/hass-victron #299: Integration stops updating entities after 1-2 days

msebald, 2025-02-20:
Hi,

I am having the issue here that the Victron integration stops updating entities after 1-2 days. After I restart HA, everything is updated again. I also tried to restart the Cerbo GX for a test, it had no effect.

No idea if I have this issue from the beginning, as I am about to migrate from openHAB to HA since the beginning of January and therefore restarted HA a lot.

I am running the latest version of the integration plus the latest Docker version of HA. And I am happy to provide more information, data and so on - I am just not sure what could lead to a better understanding what is going wrong here.

Regards

gepd, 2025-02-21:
I'm having the same behavior with my integration, everything was working before 2025.2.x

tenekes, 2025-02-21:
Same issue here after upgrading to ver. 2025.2.4

msebald, 2025-02-21:
Found something in the logs which occurs regulary when there is no more data collected until HA restart:

```
2025-02-21 23:04:20.892 ERROR (MainThread) [custom_components.victron.coordinator] Unexpected error fetching victron data
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/victron/coordinator.py", line 78, in _async_update_data
    data = await self.fetch_registers(unit, register_info_dict[name])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/victron/coordinator.py", line 179, in fetch_registers
    return await self.hass.async_add_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.api_update, unit, registerData
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/victron/coordinator.py", line 201, in api_update
    return self.api.read_holding_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        unit=unit,
        ^^^^^^^^^^
        address=self.api.get_first_register_id(registerInfo),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        count=self.api.calculate_register_count(registerInfo),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/victron/hub.py", line 40, in read_holding_registers
    return self._client.read_holding_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=slave
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 114, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=slave))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 203, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 104, in sync_execute
    self.pdu_send(request)
    ~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 160, in pdu_send
    self.low_level_send(self.trace_packet(True, packet), addr=addr)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/tcp.py", line 221, in send
    return self.socket.send(request)
           ~~~~~~~~~~~~~~~~^^^^^^^^^
BrokenPipeError: [Errno 32] Broken pipe
```

timoburggraf, 2025-02-22:
I have the same issue and the same error entry.

tenekes, 2025-02-22:
After upgrading to HA Core: 2025.2.5 seems to have fixed the issue

msebald, 2025-02-23:
Same here. A few hours after the upgrade the error is back in the log and no more data is gathered.

RichardtGit, 2025-02-23:
Having the same issue. Made an automation to update the values every 10 minutes until the issue is resolved,

alias: Reload Victron
description: ""
triggers:
  - trigger: time_pattern
    minutes: /10
conditions: []
actions:
  - data: {}
    target:
      entity_id: sensor.victron_system_battery_soc
    action: homeassistant.reload_config_entry
mode: single

sfstar, 2025-02-24:
Thanks all for providing information regarding the issues experienced.
For now you could use the automation in this comment (from a previous issue) to "fix" the integration until the next hickup.
https://github.com/sfstar/hass-victron/issues/64#issuecomment-1488023571 (It also checks if the last update was longer than a minute ago so that reloads only happen when necessary and can be checked more frequently)

I'm looking at an partial rewrite for the connection logic to make this more robust (currently the retry logic has some bugs in it).
Due to the amount of work required this might take 1 or 2 weeks.
If you notice any other behaviour related to this issue by the integration (feel free to comment on this issue.)

dmsims, 2025-03-03:
Correctly formatted yaml for reload automation

`alias: Victron reload
description: ""
triggers:
  - trigger: time_pattern
    minutes: /10
conditions: []
actions:
  - action: homeassistant.reload_config_entry
    target:
      entity_id: sensor.victron_system_battery_soc
    data: {}
mode: single
`

dmsims, 2025-03-03:
Correctly formatted code for yaml automation
```
alias: Victron reload
description: ""
triggers:
  - trigger: time_pattern
    minutes: /10
conditions: []
actions:
  - action: homeassistant.reload_config_entry
    target:
      entity_id: sensor.victron_system_battery_soc
    data: {}
mode: single
```

trentjw, 2025-05-16:
Now it will no longer update after a few hours have to reload it in home assistant every time I need to get a status report