sfstar/hass-victron #325: Switch for generator start/stop disappeared

Grisen80, 2025-04-19:
## The problem
 After upgrade to v 0.5, all entities related to my generator connected to relay 1 on my Cerbo GX have become unavailable (ex the 'switch.victron_generator_autostartenabled_0' that earlier had been autogenerated by the integration)

## Environment
Victron Cerbo Gx v3.55

- Home Assistant Core release with the issue: 2025.4.2
- Last working Home Assistant Core release (if known):
- Integration version causing this issue:  0.5.0

## Problem-relevant `configuration dialog`

```yaml

```

## Traceback/Error logs

```txt

```

## In case of unit ID issue, missing devices or non-updating devices

## Additional information

Sry if I have not filled everything out correctly but not used to submitting bug requests. Thnks for a great integration.

jwiesel, 2025-05-07:
Hi @Grisen80 ,
while going through all entities in PR #327 I realized, that there are only sensors and binary sensors in my integration.

No entities of type buttons, numbers, select or switch. - Might be a larger issue.

Grisen80, 2025-05-07:
@jwiesel, I also noted that the generator and the cerbo share the same UnitID (100 in my case). Could be this issue is related to problem w double ids as described in #326

jwiesel, 2025-05-08:
@Grisen80 : regarding your last question: I just checked my setup and don't have that situation. Therefore I am not sure.

Regarding UnitID 100 I found a commit from a previous release indicating that these devices (except settings, system) will get a device id 0:
https://github.com/sfstar/hass-victron/pull/206/files

closed 2025-08-07 by github-actions[bot]