sfstar/hass-victron #337: Victron V3.60 modbus registers out of date

ABroodryk, 2025-06-15:
## The problem

The victron modbus registers is out of date for Victron Venus OS V3.60 and error on reconfiguring the integration:

[image]

## Environment

- Home Assistant Core release with the issue: 2025.6.1
- Last working Home Assistant Core release (if known): N/A
- Integration version causing this issue: v0.5.0

## Problem-relevant `configuration dialog`

xx
```yaml

```

## Traceback/Error logs

```txt
This error originated from a custom integration.

Logger: custom_components.victron.coordinator
Source: helpers/update_coordinator.py:380
integration: victron (documentation, issues)
First occurred: 10 June 2025 at 15:02:12 (91618 occurrences)
Last logged: 22:16:57

Unexpected error fetching victron data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/victron/coordinator.py", line 79, in _async_update_data
    data = await self.fetch_registers(unit, register_info_dict[name])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/victron/coordinator.py", line 187, in fetch_registers
    return await self.hass.async_add_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.api_update, unit, registerData
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/victron/coordinator.py", line 212, in api_update
    return self.api.read_holding_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        unit=unit,
        ^^^^^^^^^^
        address=self.api.get_first_register_id(registerInfo),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        count=self.api.calculate_register_count(registerInfo),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/victron/hub.py", line 48, in read_holding_registers
    return self._client.read_holding_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=slave
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 114, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=slave))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 203, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 104, in sync_execute
    self.pdu_send(request)
    ~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 160, in pdu_send
    self.low_level_send(self.trace_packet(True, packet), addr=addr)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/tcp.py", line 221, in send
    return self.socket.send(request)
           ~~~~~~~~~~~~~~~~^^^^^^^^^
BrokenPipeError: [Errno 32] Broken pipe
```

## In case of unit ID issue, missing devices or non-updating devices

[image]

[image]

[image]

[image]

## Additional information

21-Jailbraik-21, 2025-06-20:
Habe heute alles was benötigt wird über MQTT eingebunden... Und zum Glück bleiben die Statistiken erhalten.... 

Anwendung jetzt gelöscht.... SCHADE, aber wenn es nicht gepflegt wird, kostet es nur Nerven.

janCGN, 2025-06-21:
Ich hab den Umzug auf mqtt auch erfolgreich vollzogen.  Läuft bis jetzt super. Viele Grüße

janCGN, 2025-06-21:
Kann folgendes empfehlen: https://github.com/evenssolutions/hass-victron-mqtt-discovery

Uhae, 2025-06-26:
@21-Jailbraik-21 Könntest Du vielleicht kurz erläutern, wie Du den Umzug auf MQTT angegangen bist? Ich habe auch Sorgen, dass die ganzen Statistiken, insbesondere im Energy Dashboard, dann nicht mehr funktionieren wenn ich "neue" Entities erstelle.

Danke!

21-Jailbraik-21, 2025-06-26:
War eigentlich relativ einfach, aber mit Home Assistant über den Mqtt Explorer auf die Vectron Daten zugegriffen beziehungsweise abgefragt und diese dann in Home Assistant integriert. Wenn ich die Entitäten ausgetauscht habe, wurden die alten Historie nicht gelöscht. Das war mir auch neu.

Funktioniert bis jetzt super kann leider nicht installierte antworten, weil ich noch auf der Arbeit bin   
Hoffe das reicht erst mal

janCGN, 2025-06-26:
Guck dir mal das Addon an, was ich geoostet habe. Erspart einiges an Arbeit

21-Jailbraik-21, 2025-06-26:
> Guck dir mal das Addon an, was ich gepostet habe. Erspart einiges an Arbeit

WOW - das ist ja krass - alle Daten ohne eine Config ... COOL. Danke...
Schau ich mir noch genauer an.

Hoffe das wird gut gepflegt.

DANKE