sfstar/hass-victron #351: Stops pulling data

andrewheathfield17-creator, 2025-08-05:
All entities etc work 100%

2 or 3 times a day it just freezes when pulling info. 

I will then need to reload the integration and then it works perfectly again. 

This is with the latest update... Not sure what the issue is.

dtrv, 2025-08-20:
same here but less frequent.
appears afaics every two days.
after reloading everything works again.

JohnMcLear, 2025-11-18:
I'm getting this every time my network access to the unit becomes unavailable.

```
Unexpected error fetching victron data

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 403, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/victron/coordinator.py", line 77, in _async_update_data
    data = await self.fetch_registers(unit, register_info_dict[name])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/victron/coordinator.py", line 179, in fetch_registers
    return await self.hass.async_add_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.api_update, unit, registerData
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/victron/coordinator.py", line 204, in api_update
    return self.api.read_holding_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        unit=unit,
        ^^^^^^^^^^
        address=self.api.get_first_register_id(registerInfo),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        count=self.api.calculate_register_count(registerInfo),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/victron/hub.py", line 88, in read_holding_registers
    return self._client.read_holding_registers(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, device_id=slave
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 113, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=device_id))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 202, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 128, in sync_execute
    self.pdu_send(request)
    ~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 211, in pdu_send
    self.low_level_send(self.trace_packet(True, packet), addr=addr)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/tcp.py", line 221, in send
    return self.socket.send(request)
           ~~~~~~~~~~~~~~~~^^^^^^^^^
BrokenPipeError: [Errno 32] Broken pipe
```

It never attempts to reconnect/reload itself.