sfstar/hass-victron #360: v0.7.0 pre release (please read release notes before updating)

sfstar, 2025-08-26:
## The problem

This is an issue that will be used as a thread to index what issues are found with the new register set added in the 0.7.0-beta release.

Link to release notes:
https://github.com/sfstar/hass-victron/releases/tag/v0.7.0-beta

## Environment

- Home Assistant Core release with the issue:
- Last working Home Assistant Core release (if known):
- Integration version causing this issue:

## Problem-relevant `configuration dialog`

```yaml

```

## Traceback/Error logs

```txt

```

## In case of unit ID issue, missing devices or non-updating devices

## Additional information

kucerazdenek, 2025-08-27:
victron venus os version v3.55
HA container (Synology) - 2025.7.2
3-phase system (3x 25A), Fronius 8.2.3-M, 3x Victron Multiplus II/3000, 6x Pylontech US5000
I have seen battery device on ID 225 in v0.6.1, all entities were available.
After upgrade to 0.7.0 all entities under ID 225 got unavailable.
After rescan, device ID 225 disappeared as such

[image]

[image]

sfstar, 2025-08-31:
Thank you for testing, due to upcoming breaking changes in the new 2025.9 release as discovered by @jacauc in #361 I will revert the changes made by the beta release in order to release an hotfix without these changes included.
Once done, the changes will be re-applied and 0.7.0-beta re-released.
Please note that the battery detection issue as of yet hasn't been resolved

sfstar, 2025-08-31:
#366 will reintroduce the 0.7.0 functionality and will be remerged at 9th of september the earliest.
To allow for any other issues with the upcoming 2025.9 release to be addressed before these breaking changes are reintroduced.

VibroAxe, 2025-09-20:
> > [#366](https://github.com/sfstar/hass-victron/pull/366) will reintroduce the 0.7.0 functionality and will be remerged at 9th of september the earliest. To allow for any other issues with the upcoming 2025.9 release to be addressed before these breaking changes are reintroduced.
> 
> @sfstar I really appreciate the hard work you do for all of us. May I ask you if there is some schedule/estimate, when version working with Venus OS/Cerbo GX 3.60+ will be released? In 3.60, Victron improved DynamicESS algorithm and this functionality finally become usable, but because of changes to modbus registers I cannot upgrade to this version and still stuck on 3.55. Thanks.

I just hit the same problem that my cerbo is on 3.70 and home assistant is on 2025.9 causing this integration to get stuck in the middle as the upgraded register set doesn't exist

kucerazdenek, 2025-10-28:
Still same error on 0.7.0-beta-2 with Pylontech on ID 225 as decribed at beginning of this thread.

kucerazdenek, 2025-11-01:
> Still same error on 0.7.0-beta-2 with Pylontech on ID 225 as decribed at beginning of this thread.

After upgrade of Venus OS from version 3.55 to 3.66, ID 225 battery reappeared.

sfstar, 2025-11-17:
v0.7.0-beta-5 has been made (currently running on my production instance without issue)

kucerazdenek, 2025-11-17:
Thanks, works without issues so far on my test HA instance (2025.11.2 + Venus OS 3.66)
beta-4 - 253 entites in my configuration
beta-5 - 323 entites

closed 2026-03-22 by github-actions[bot]