sfstar/hass-victron #361: No compatibility with PyModbus of 2025.9.b0

jacauc, 2025-08-28:
This is probably known, but I did not find any existing issue for this.
Integration fails to startup with HA 2025.9.b0 likely due to PyModbus API changes.
`Failed setup, will retry: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'`

## Environment

- Home Assistant Core release with the issue: 2025.9.b0
- Last working Home Assistant Core release (if known): 2025.8.3
- Integration version causing this issue: 0.6.1

## Problem-relevant `configuration dialog`

```yaml

```

## Traceback/Error logs

```txt

```

## In case of unit ID issue, missing devices or non-updating devices

## Additional information

roberto-Jade, 2025-08-29:
I had to change hub.py:
return self._client.write_register(address=address, value=value, slave=slave)
To:
return self._client.write_register(address=address, value=value, device_id=slave)

And this:
return self._client.read_holding_registers(
    address=address, count=count, slave=slave
)
To:
return self._client.read_holding_registers(
    address=address, count=count, device_id=slave
)

jacauc, 2025-08-30:
Yep, that's exactly what my PR does which is linked above

tpomerhn, 2025-09-03:
Okay thanks, looks like 9.0 is accessible now in HA, updating now. Guess I was just in that limbo between 8.3 being the latest and 9.0's release, when 0.6.2 was the latest in the Victron integration. Looks like it's working now on 9.0 for me.

Would be good if in the future this kinda thing happens to put in an if/elif/else or try/except loop to handle versioning, if the value is handled the same but just named differently. But no matter now, it's working. Thanks for your work on this!

jacauc, 2025-09-04:
I suspect it's simply due to infrequent update refresh frequency of HA itself

kucerazdenek, 2025-09-04:
Works like a charm. Even there is possible to upgrade Victron integration on HA version lower than 2025.9, I did upgrade of HA to 2025.09 first and then victron integration to 0.6.2

sfstar, 2025-09-04:
If the update isn't coming through you can install it via hacs (as you would normally do for new integrations).
By selecting a specific release when prompted and installing 0.6.2.